Maritime-Robotics-Student-Society/Boat-construction #30: Purchase waterproofing materials

smaria, 2016-07-11:
We need to buy some waterproofing materials, I would propose plasti dip and silicone:
https://www.amazon.co.uk/Plasti-Dip-Plastic-Rubber-Paint/dp/B0114Z42F8/ref=sr_1_7?ie=UTF8&qid=1468225472&sr=8-7&keywords=plasti+dip 

https://www.amazon.co.uk/Everbuild-EVBGPSESQCL-General-Purpose-Silicone/dp/B001V9NANS/ref=sr_1_10?ie=UTF8&qid=1468225658&sr=8-10&keywords=silicone 

Plasti dip is for covering entire parts, silicone for more selective work.
It is probably better to buy several smaller packages of silicone since they will be easier to apply.
Any other opinions?

smaria, 2016-07-11:
We probably also need something to waterproof the servos that will allow them to keep moving - some sort of oil?

Nanoseb, 2016-07-11:
I thought we already bought some marine grease to waterproof servos.

smaria, 2016-07-11:
that is correct, I saw that in our purchase list just ten minutes before writing this ticket... thanks

closed 2016-11-26 by smaria