Maritime-Robotics-Student-Society/Boat-construction #33: Make overview of waterproof servos

smaria, 2016-11-24:
Because IP68 / IP67 servos will not break that quickly

Nanoseb, 2016-11-24:
Graupner has some sail winches that are 'splashproof', they don't mention any IP spec but it is probably better than the one we already have.
https://www.graupner.com/RC-Electronics/Servos/Sailwinch

smaria, 2016-11-26:
I just asked these for datasheets and a quote: 
http://www.jvl.dk/808/integrated-servo-motors

smaria, 2016-11-26:
Nevermind, found the datasheed; the smallest is still > 2.5 kg. Not for this boat ;)

smaria, 2016-11-26:
https://www.sparkfun.com/products/11886
IP 67
$ 43
180 degree rotation

Nanoseb, 2016-11-26:
http://www.emaxmodel.com/servos/waterproof-servo/es3005-waterproof-servo.html
'waterproof' but they don't mention any certification.
~$15

Nanoseb, 2016-11-26:
http://www.arrma-rc.com/part/AR390133
£17