Maritime-Robotics-Student-Society/Boat-construction #34: Make a weights table

smaria, 2016-11-24:
Batteries
Servos
Wires
Keel (-> Length vs Weight)

smaria, 2016-11-24:
And there may be even more missing ;)

smaria, 2016-12-15:
any updates on this? Could you add the ongoing work table here or in the google drive?

TAJD, 2016-12-19:
Apologies, no updates on this. Would it be possible to meet up during the week just to dig all the components out for me to go and measure? I find evenings challenging, but I don't think I'd need more than 5 mins of time to locate all the items.

smaria, 2017-01-06:
I think you will need at least 30 minutes, I propose you come over during lunch time/after the FSI seminar. We have a scale in the lab at the bottom which I can borrow, and a caliper. Ideally we should have a list of all parts and what measurements are needed beforehand, otherwise it will be too easy to miss some bits.

TAJD, 2017-01-07:
Sounds like a plan! I'll get on with making a list then.