Maritime-Robotics-Student-Society/Boat-construction #40: Study how weight and length of keel influence the stability of the boat

tsaoyu, 2016-11-24:
Using software like MaxSurf or excel sheet with CAD drawings.

tsaoyu, 2016-11-24:
Related to: https://github.com/Maritime-Robotics-Student-Society/Society-materials/issues/36

Nanoseb, 2016-11-24:
if needed I have access to Maat hydro here, I can run some stability calculations

TAJD, 2016-11-25:
Do we have an existing CAD model or weight book? If we don't then I think we need to focus on creating these before doing any basic design work.

pierbiag, 2016-11-25:
We have nothing for the weight, however I have a relatively rough linesplan in autocad for the other boat 

Pier Maria Biagiolini

TAJD, 2016-11-25:
Ok, I can get on with recording the weights on the boat. Would it be possible if you could upload the lines plan you currently have?

smaria, 2016-12-15:
The linesplan is in the sailing robot google drive, let me know if you don't have access to it yet